schote/nexus-console #30: BUG: Enabling more then two channels is not possible

schote, 2023-10-28:
- When more than two channels are enabled in the device configuration, the read number of enabled channels from card equals zero 
- Happens directly after channel enable: `sp.spcm_dwSetParam_i32(self.card, sp.SPC_CHENABLE, channel_selection)`
- Enabling channel 0 and/or channel 1 works, even independently

### Resolution attempts
- Tried to replaced `channel_selection` by `sp.CHANNEL0 | sp.CHANNEL1 | sp.CHANNEL2` and observed the same behavior
- Tried to set the channel amplitude to `10000` according to the other channels

hiddendb, 2023-11-16:
The only valid number of active channels is 2^n. Activating 4 channels is working without a problem.

schote, 2023-11-16:
> The only valid number of active channels is 2^n. Activating 4 channels is working without a problem.

Thanks for figuring out! 
Did you also try 8 channels?
Do you remember the section in the manual? We should reference it somewhere in the code.

hiddendb, 2023-11-16:
- 8 channels is working as well. Can get a noise signal out of all of them.
- Reference in manual is p. 72 [https://spectrum-instrumentation.com/dl/m2p_59xx_manual_english.pdf]
- `device_config.yaml`should be a good place

PR #35 (2023-11-16): Image Acquisition Leiden
- [x] Implementation of TSE Sequence 2D/3D
- [x] Update gradient scaling by hardware parameter: coil sensitivity, gpa gain
- [x] Add image reconstruction algorithm
- [x] Update sequence plot
- [x] Improvement of rollout performance
- [x] Switch to ruff for linting
- [x] Enable execution of experiment scripts from console

closes #34 
closes #32 
closes #28 
closes #26 
closes #36 
closes #30
closed 2023-11-16 by schote