schote/nexus-console #57: Sequence fails to execure fully, all events happen at once?

LUMC-LowFieldMRI, 2024-02-15:
During tuning of the gradient correction parameter occasionally the sequence would fail to run to it;'s full duration and it appeared like all (most?) steps happened instantaneously.

When executing the sequence again a critical error would be thrown (failed to record what it was).

Appears like this happens randomly, re-running with the same sequence parameters does not trigger the same behavior.

schote, 2024-03-12:
Could it be that the problem mainly occurs when repeating long sequences? At least that is the only case where I have observed this problem. I have never had this problem with short sequences, i.e. SE spectrum sequence for instance.

Toreil, 2025-03-13:
Appears to be solved with the new sequence unroller

closed 2025-03-13 by Toreil