schote/nexus-console #58: Revision of gradient offsets implementation

schote, 2024-02-26:
Currently, a gradient offset is generated by the interpreter, who sets the offset value instead of a zero as the base level of the unrolled sequence. However, as described in the spectrum-card manual, there is also the option to set an offset value (see p. 71).

### Revision

Registers for output offsets: `SPC_OFFS1` - `SPC_OFFS3` for x, y and z gradients. Should be set by `TXDevice` according to the offset values in acquisition parameters. 
Offset value must be considered when checking the maximum amplitude, because offset + amplitude must not exceed the maximum output voltage (12V for high impedance, 6V for 50 ohms).

LUMC-LowFieldMRI, 2024-02-27:
How/where is the shim offset currently defined? Can the value be written?

schote, 2024-02-27:
> How/where is the shim offset currently defined? Can the value be written?

Currently, the shim offsets are defined as gradient offsets within the [acquisition parameters](https://github.com/schote/spectrum-console/blob/main/src/console/spcm_control/interface_acquisition_parameter.py).
Currently, the acquisition parameters are set in the acquisition control, but as discussed, they might become a global variable as they also contain the larmor frequency and b1-scaling.

LUMC-LowFieldMRI, 2024-02-28:
Could you give some demo code for that? Thanks!

schote, 2024-02-28:
> Could you give some demo code for that? Thanks!

With the latest pull-request, the acquisition parameters become a global object which can be saved and loaded using a pickle file.
In general, the acquisition parameters can be defined as follows:
```
parameters =  AcquisitionParameter(
    larmor_frequency=2e6,
    b1_scaling=1.0,
    gradient_offset=Dimensions(x=0, y=0, z=0),
    fov_scaling=Dimensions(x=1., y=1., z=1.),
    decimation=200,
    num_averages=1,
    averaging_delay=0.0
```
Before the execution of an unrolled sequence is started, the gradient offsets are set by the acquisition control within the `run()` method.
```
self.tx_card.set_gradient_offsets(parameter.gradient_offset)
```
After the acquisition, the offset is reset to 0 for each of the gradient channels.

schote, 2024-02-28:
> Thanks! and the units are?

The units are in mV, I added this information to the docstring.

PR #59 (2024-03-05): Fixes and improvements from in-vivo session
This pull-request gathers multiple fixes and improvements to the console application, which were applied during an in-vivo imaging session at LUMC. The commits were cherry-picked from the lumc-experiments branch and only contain modifications to the package `src` directory.

### (Quick-)Fixes

- Update on 3D TSE sequence: Fixed timing, trajectory calculation, channel orientation
- Update on calibration sequences
- Fixed gate duration rounding issue in sequence provider (lead to missing sample points)
- Fixed saving of acquisition data: Missing averag [... cut]
closed 2024-03-05 by schote