schote/nexus-console #63: Cannot acquire data with 4 channels simultaneously

LUMC-LowFieldMRI, 2024-03-20:
I can run the TSE that i've implemented with two receive channels simultaneously but when I increase the channel count to 4 the sequence crashes after the first echo train with the following error:

Exception in thread Thread-5 (_gated_timestamps_stream):
Traceback (most recent call last):
  File "c:\Users\Tom\code\envs\console\lib\threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "c:\Users\Tom\code\envs\console\lib\site-packages\ipykernel\ipkernel.py", line 761, in run_closure
    _threading_Thread_run(self)
  File "c:\Users\Tom\code\envs\console\lib\threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Tom\code\spectrum-console\src\console\spcm_control\rx_device.py", line 415, in _gated_timestamps_stream
    gate_data = np.ctypeslib.as_array(ptr_to_slice, ((total_bytes // 2),))
  File "c:\Users\Tom\code\envs\console\lib\site-packages\numpy\ctypeslib.py", line 524, in as_array
    p_arr_type = ctypes.POINTER(_ctype_ndarray(obj._type_, shape))
  File "c:\Users\Tom\code\envs\console\lib\site-packages\numpy\ctypeslib.py", line 357, in _ctype_ndarray
    element_type = dim * element_type
ValueError: Array length must be >= 0, not -809199968

the sequence keeps running after the error is thrown but only one line of k-space is non-zero once the data is sorted/reconstructed

berksilemek, 2024-03-26:
The issue was successfully replicated in Berlin console , pinpointing the rx_device module (spectrum-console/src/console/spcm_control/rx_device.py) as the source. In the context of spectrum console's receive cards, there's a minimum page size requirement—currently set at 4KB—which also matches the receive notify size. Pages that aren't fully populated cannot be accessed by the application, necessitating careful data management by the user.

The total number of bytes per gate is calculated as follows:
```python
   total_bytes = (gate_sample + self.pre_trigger) * 2 * self.num_channels.value
```
In this equation, gate_sample is determined based on time samples, and self.pre_trigger is utilized to ensure alignment with the correct samples. Subsequently, we compute the leftover bytes and update the read operation on the spectrum cards accordingly:
```python
# It is likely that the entire page isn't filled, leaving some bytes in the buffer unreadable.
bytes_leftover = (total_bytes + self.post_trigger_size) % rx_notify.value

# We then tally these leftover bytes. If this sum exceeds the notify value, we proceed to read the page.
total_leftover += bytes_leftover
if total_leftover >= rx_notify.value:
    total_leftover = total_leftover - rx_notify.value
    available_card_len = bytes_sequence - (bytes_leftover) + rx_notify.value
else:
    available_card_len = bytes_sequence - (bytes_leftover)
```
The computation of available_card_len is vital for ensuring synchronization between the application and device buffers. The initial calculation of bytes_leftover failed to consider the correct size because self.post_trigger_size depends on the number of channels. The correction is as follows:
```python
 # It is likely that the entire page isn't filled, leaving some bytes in the buffer unreadable.
bytes_leftover = (total_bytes + self.post_trigger_size*self.num_channels.value) % rx_notify.value
```
This fix will be merged to the main.

LUMC-LowFieldMRI, 2024-03-27:
Will be closed automatically by PR

closed 2024-05-17 by schote