schote/nexus-console #66: Phase instability during and between acquisitions

LUMC-LowFieldMRI, 2024-03-28:
There seem to be some phase stability issues with the data acquisition. For B0 mapping images with two slightly different timings are acquired and the phase difference between the two acquisitions is directly proportional to the local B0 offset. In order for this to work the phase has to be fully stable otherwise the instability results in errors in the B0 map.

The two images are acquired using a single seq file to maximise stability but there are still substantial errors in the B0 map, and if the same sequence is ran twice wildly different results can be obtained, this indicates that there is a problem with the phase not being consistent in the system. I've attached two acquisitions to demonstrate the issue
[image]
[image]

berksilemek, 2024-03-31:
In general, we (@schote) need to test and report phase stability of the console. The internal clock has its limitations. We need to determine and report our limitations for the application. 
@LUMC-LowFieldMRI What were the tested TR values? Total acquisition times? Did you use the updated rxdevice.py?

schote, 2024-04-26:
I think a potential issue is the phase correction step, that we apply when the data is demodulated and decimated.
We should investigate if this phase manipulation step is really necessary and if it is correct.

LUMC-LowFieldMRI, 2024-04-27:
We tried a few things already, including no phase correction and phase correction before decimation (which I think is actually how it should be). No phase correction was a disaster and shows that clearly the phase correction is needed. The correction before the decimation also didn’t work but that might just have been a bug in the implementation.

Another thing to check is to see if moving from the FIFO acquisition to a triggered acquisition. If you read the description of the starhub (which I know you don’t have on the system in Berlin) it sounds like this is designed to take handle phase sensitive acquisitions.

We should investigate if this phase manipulation step is really necessary and if it is correct @berksilemek<https://github.com/berksilemek>.

Toreil, 2025-03-06:
I did some more digging in to this and was able to acquire images without needing to do the phase correction but would intermittently end up with some ghosting in the image so it may not be (fully) related to the phase correction.

Toreil, 2025-03-15:
The top images are acquired without the reference phase, the bottom ones with. The top images are sharper which we expect since they have the correct phase relationship between all the RF pulses. Both images show ghosting in the phase encoding direction which I think comes from 'normal' MR physics issues, not any phase instability.

[image]
[image]
[image]
[image]

berksilemek, 2025-03-18:
> The top images are acquired without the reference phase, the bottom ones with. The top images are sharper which we expect since they have the correct phase relationship between all the RF pulses. [...]

Hi Tom, 

Thanks for the analysis and the images! I don’t have the complete system details, such as whether the Spectrum card’s internal oscillator is used, if a 10 MHz reference clock is connected between Tx and Rx, or if the Star-Hub is installed and enabled. However, phase stability still could be a concern here. While MR physics can introduce phase-related artifacts, my gut feeling is that hardware synchronization may still play a role. If the clock phase drifts significantly per TR, it can manifest as ghosting in the phase-encoding direction. The best way to isolate this is to control for clock stability and sequence timing.

Clock Stability Considerations
From the datasheet, the Spectrum card’s internal clock accuracy is ±1.0 ppm, meaning that at 2 MHz, the signal could drift by ±2 Hz. In comparison, high-field MRI scanners typically use Oven-Controlled Crystal Oscillators (OCXOs) with ±0.01 ppm or better to maintain phase coherence.
Even if we ignore jitter and other noise sources, a 1 ppm drift results in a phase shift of 360° per TR (500 ms). While the average frequency drift cancels out over long scans, the accumulated phase shift per TR could still contribute to ghosting artifacts.
Potential Approaches to Investigate Further
1.	Improve Clock Stability
o	If possible, supply a more stable external 10 MHz reference clock (or other) instead of using the Spectrum card’s internal oscillator.
o	Check if the Star-Hub is used for synchronization—this could help with phase alignment.
2.	Adjust the Sequence Timing
o	Shorten TR or sequence duration to see if ghosting reduces. If ghosting improves, it suggests a clock-related issue.
o	Prolong the sequence duration to see if any frequency-dependent effects become noticeable.