schote/nexus-console #70: Long calculation time for decimation

schote, 2024-05-16:
Problem:
We use the scipy decimate function with the FIR filter to decimate our acquired signals. The calculation takes a very long time for 3D volume imaging. 

Suggestions:
Improve the decimation speed by using an FFT based method

Toreil, 2024-11-12:
I looked into this a bit and found that the performance difference using FFT vs the scipy decimation method is minimal (likely since I believe the scipy implementation of the decimation uses an FFT at the backend). There is a small benefit to be had due to the fact that using the FFT you can combine both the demodulation and decimation but the decimation loses a little accuracy since it's limited to the frequency resolution of the acquisition.

Toreil, 2025-07-02:
Calculation speed is improved with parallel processing in PR #94, specifically see the comment about the speedup offered by multithreading:

https://github.com/schote/nexus-console/pull/94#issuecomment-3018663447

closed 2025-07-02 by Toreil