schote/nexus-console #80: Datatype of dimensions in ISMRMRD export

schote, 2025-03-03:
The dimensions are exported as strings, but integer values should be used

rkcatarina, 2025-03-24:
Here are some other suggestions for modifiying the ISMRMRD export:
- ECO labels ("contrast") have to be set
- Aquisition times have to be set
- Reconstruction matrix size and encoding matrix size should be integer (I think it is similar to the issue that @schote already mentioned three weeks ago above)

rkcatarina, 2025-06-25:
There are still some issues when reading in the ISMRMRD-header:

- **header.encoding[0].reconSpace.matrixSize.x** is apparently a string, but should be an integer (same for **y** and **z**)
- **header.encoding[0].encodedSpace.matrixSize.x** is also apparently a string, but should be an integer (same for **y** and **z**)
- contrast and reptitions labels are not set in header acquisitions (something like **acq.idx.contrast** and **acq.idx.repetition**)

rkcatarina, 2025-07-02:
[image] 

I am trying to reconstruct the reference sequences and realized that some additional parameters are not correctly saved in the header. Among other things TE, TR,TI and FA are not correctly indicated, as you can see in the screenshot. Maybe we could discuss which of the other parameters (vendor etc.) would be relevant.

schote, 2025-07-14:
> I am trying to reconstruct the reference sequences and realized that some additional parameters are not correctly saved in the header. Among other things TE, TR,TI and FA are not correctly [...]

Parameters such as TE, TI, etc., must be set during header generation in conjunction with the sequence, and cannot be handled within the console code.

schote, 2025-07-21:
> There are still some issues when reading in the ISMRMRD-header:
>
> * **header.encoding[0].reconSpace.matrixSize.x** is apparently a string, but should be an integer (same for **y** and **z**)
> * [...]

The encodings need to be defined in the header which is handed to the `export_ismrmrd()` method. I found that the corrupted values   are floats stored as strings, but I think integers are expected. It could be, that these values are stored as strings, because floats where assigned. Please verify the header generation.