schote/nexus-console #82: Missing samples in acquisition on first run.

Toreil, 2025-03-13:
When running the TSE example sequence with a low number of phase encoding steps (15x10) the Rx card doesn't get all of the data points that it expects (usually one ETL fewer than it should i.e. 93 instead of 100 when the ETL is 7). On subsequent runs, without restarting the acquisition controller, this doesn't happen. This issue also only started happening with the new sequence unroller. 

It appears then that something is not done initialising (on the Rx side?) before the sequence starts to play out. I guess the best way to handle this is to add a halt before the sequence starts to play out until all systems are confirmed to be up and running.

schote, 2025-03-13:
Could you check, if the error persists with PR https://github.com/schote/nexus-console/pull/78 (and maybe add your review)? I fixes some issues with the card memory after reading the data, in some cases invalid registers were written. Maybe this  is related.

schote, 2025-05-21:
@Toreil does this issue still persist?

closed 2025-07-02 by Toreil