schote/nexus-console #90: Different BW settings in sequence lead to error in data acquisition.

h3lg3, 2025-04-28:
Trying different ADC BW using the tse_3d sequence lead to data inconsistencies when executing the sequence using the console. 
BW 20 kHz and 10 kHz worked fine, however 12, 16, 32 kHz failed. Currently I only have a screenshot of the error available. 

[image]

schote, 2025-04-28:
The problem is caused by the reference phase signal, I think @Toreil is already working on that. Currently, decimation of the MR signal is done by FIR filter and decimation of the phase reference signal is done by moving average filter. Somehow, the moving average filter ends up in a different number of decimated sample points (maybe because of a rounding issue). This causes problems when multiplying the two signals for correction.

schote, 2025-04-29:
As stated above, the issue is related to the different decimation filters used for the MR signal and the phase reference signal.
The moving average filter resulted in a different number of sampling points.
In PR #78 I added a test for a range of different decimation factors (see [commit](https://github.com/schote/nexus-console/pull/78/commits/c18bb8033f23469499b4910336f3e0931ad745f3)) which ensures, that FIR and moving average filter result in equal number of decimated sample points.

Instead of using `round(...)` to calculate the number of decimated sample points now `np.ceil(...)` is used, what results in equal shapes after decimation considering moving average and FIR filter.

schote, 2025-04-30:
@h3lg3 the sequence you provided can be executed sucessfully with the latest changes of the branch [acquisition-manager](https://github.com/schote/nexus-console/tree/acquisition-manager):

[image]

closed 2025-05-02 by schote