schote/nexus-console #91: High resolution sequence fails.

h3lg3, 2025-04-28:
Running a sequence with large matrix size (123x123x123) using the acquisition manager fails. Below is the output. 

```
Sequence name: tse_3d
Sequence duration is: 443 s
An error occurred:
File "<ipython-input-11-aaf968140abd>", line 53, in <module>
acq_data: AcquisitionData = [m.acquisition.run](http://m.acquisition.run/)()
File "<string>", line 2, in run
File "/home/guest/code/envs/console/lib/python3.10/multiprocessing/[managers.py](http://managers.py/)", line 818, in _callmethod
kind, result = conn.recv()
File "/home/guest/code/envs/console/lib/python3.10/multiprocessing/[connection.py](http://connection.py/)", line 250, in recv
buf = self._recv_bytes()
File "/home/guest/code/envs/console/lib/python3.10/multiprocessing/[connection.py](http://connection.py/)", line 414, in _recv_bytes
buf = self._recv(4)
File "/home/guest/code/envs/console/lib/python3.10/multiprocessing/[connection.py](http://connection.py/)", line 383, in _recv
raise EOFError

EOFError Traceback (most recent call last)
Cell In[11], line 53
51 with AcquisitionControlManager() as m:
52 m.acquisition.set_sequence(sequence=seq, parameter=console.parameter)
---> 53 acq_data: AcquisitionData = [m.acquisition.run](http://m.acquisition.run/)()
55 # # Calculate sequence and perform acquisition
56 # acq.set_sequence(sequence=seq)
57
58 # # # Execute the sequence and sort kspace array
59 # acq_data: AcquisitionData = [acq.run](http://acq.run/)()
61 if SAVE_RECO:

File <string>:2, in run(self, *args, **kwds)

File ~/code/envs/console/lib/python3.10/multiprocessing/[managers.py:818](http://managers.py:818/), in BaseProxy._callmethod(self, methodname, args, kwds)
815 conn = self._tls.connection
817 conn.send((self._id, methodname, args, kwds))
--> 818 kind, result = conn.recv()
820 if kind == '#RETURN':
821 return result

File ~/code/envs/console/lib/python3.10/multiprocessing/[connection.py:250](http://connection.py:250/), in _ConnectionBase.recv(self)
248 self._check_closed()
249 self._check_readable()
--> 250 buf = self._recv_bytes()
251 return _ForkingPickler.loads(buf.getbuffer())

File ~/code/envs/console/lib/python3.10/multiprocessing/[connection.py:414](http://connection.py:414/), in Connection._recv_bytes(self, maxsize)
413 def _recv_bytes(self, maxsize=None):
--> 414 buf = self._recv(4)
415 size, = struct.unpack("!i", buf.getvalue())
416 if size == -1:

File ~/code/envs/console/lib/python3.10/multiprocessing/[connection.py:383](http://connection.py:383/), in Connection._recv(self, size, read)
381 if n == 0:
382 if remaining == size:
--> 383 raise EOFError
384 else:
385 raise OSError("got end of file during message")

EOFError:

```

Toreil, 2025-04-28:
Are you on the acquisition manager branch?

h3lg3, 2025-04-28:
Yes, I am on the acquisition manager branch. The sequence which results in the error can be found here:

[tse_3d_20250422_TR600_TE12_FOV246_123px_ETL16.txt](https://github.com/user-attachments/files/19941736/tse_3d_20250422_TR600_TE12_FOV246_123px_ETL16.txt)

schote, 2025-04-28:
Could be related to the larger amount of data which is returned through the sub-process connection. If this is an issue, we could think about writing the data to a temporary file, already during the acquisition. This would also ensure, that no data gets lost if the execution fails.

To circumvent this issue, acquisitions can still be done without the use of sub-processes even in the acquisition-manager branch, see:
https://github.com/schote/nexus-console/blob/3e71b2a2286f15713ff95ed257dc691f7ceb1401/examples/tse_3d.py#L38

schote, 2025-04-30:
When using the acquisition manager now shared memory is used to transfer the data between the processes. When returning the acquisition data to the other process as it is, the object is pickled and transferred. This appears to be challenging for larger amount of data. Thus now shared memory is created and only the reference is returned within the acquisition data. In the process which receives the data, the array is re-created from shared memory. This works also for the sequence @h3lg3 uploaded here:

[image]

[image]

schote, 2025-04-30:
The use of shared memory in interactive windows of vs code sometimes causes the kernel to crash. Thus this issue needs further investigation. I think the overall acquisition data handling deserves a revisioning process, which is why I would like to postpone this issue as it only affects largeer amounts of data.

Toreil, 2025-07-17:
Seems to be at least somewhat related to OS since running it with the following settings works on Ubuntu but didn't work on Windows:

Toreil, 2025-07-23:
It appears that this is related to an issue with reading data from the cards rather than the acquisition manager itself and will be addressed in a PR shortly.