schote/nexus-console #98: Updating console branch leads to image artifact

h3lg3, 2025-06-20:
Hi, 
I just updated my local console branch from "acquisition-control manager" branch (https://github.com/schote/nexus-console/pull/78) to main (https://github.com/schote/nexus-console/commit/32b440184a1f9cb1b9f77bdd11d9f4aa85c6f0aa) and the quality control noise measurement revealed a new image artifact at the center of the image. 

**Branch: acquisition-control manager**
[image]

**Branch: main**
[image]

This is the sequence I used (noise scan, RF pulse is deactivated by setting the flip angle to 0)

[tse_3d_QA_SPHERE_NOISE_TR600_TE10_FOV140_70px_ETL50_BW20_64s_2mm.txt](https://github.com/user-attachments/files/20835881/tse_3d_QA_SPHERE_NOISE_TR600_TE10_FOV140_70px_ETL50_BW20_64s_2mm.txt)

Any ideas? Is this expected or do I need to update my code for data reconstruction?

```
decimation = int(20e6 / ro_bandwidth)
console.parameter.decimation = decimation

# Open acquisition control manager
with AcquisitionControlManager() as m:
    m.acquisition.set_sequence(sequence=seq, parameter=console.parameter)
    acq_data: AcquisitionData = m.acquisition.run()
```

h3lg3, 2025-06-20:
When I went back to my local branch of "acquisition-control manager" the artefact disappeared again. I can try with different sequence settings..

schote, 2025-06-25:
@h3lg3 to which commit are you referring exactly? There have been a couple of branches in between, I think it would help to narrow down the issue.
This is the last commit to main before the PR recently merged:
https://github.com/schote/nexus-console/commit/a7b049ecf3b593818e1379e9e18412f05c759b8a

h3lg3, 2025-06-25:
Quick update:
1) I can only check the exact commit next week, but as discussed above, there is an artifact in both versions, it just looks different. In both cases I just the acquisition manage method. 
2) The artifact is probably related to an DC offset somewhere in the receive chain. One option is that the ADC induces the offset and just needs to be re-calibrated. @schote 
3) The artifact is present even without a phantom, RFPA turned off and RF-PREAMP turned off @Toreil 
4) Increasing the echo time did not solve the issue @Toreil 
5) I can collect unprocessed data next week to further investigate.

schote, 2025-07-01:
@Toreil isn't there also a barely visible horizontal line in the image you acquired with the sequence from @h3lg3 (I am referring to the far right column, 4th row)?
I could reproduce the artifact, it is visible also when disconnecting the RF coil.
Also, it appears with the latest main branch and with commit a7b049e, which is the state before the decimation and demodulation fix.
I think the branch @h3lg3 is referring to doesn't include the phase-reference removal.
Commit 9e7fd4f yields a different result for the provided test sequence, but I don't see a center point.
In any case, the unprocessed signal looks as follows:

[image]

schote, 2025-07-01:
> That looks like what I see then, right? Do you still see the faint line in the 2D sequence?

No

closed 2025-10-06 by h3lg3