schote/nexus-console #104: Image artifact  for sequence with large 3D matrix and low readout BW

h3lg3, 2025-07-04:
Hi! 
Since a few weeks I observe an image artifact for certain sequence parameters. Mainly a large matrix size (70x70x70) combined with a low readout bandwidth (10 kHz). Weirdly the artifact disappears if I increase the bandwidth to 20 kHz (everything else unchanged). It also disappears if I change the matrix size to 70x70x10 with BW = 10 kHz. Therefore I dont think its a sequence issue.
I dont think its an eddy current issue either because it is also present with with longer TE. Also present with BW 12 kHz. Not present with BW 15 kHz or 20 kHz. Not present with 30x30x30 matrix size.  
In order to rule out a local hardware issues, I would be happy if someone could test the sequence on a different system. I attached the sequence as a .txt file. I tested the sequence on the current main branch c17fd2865e9bd918ac85b78b66e71eb3f08d70dc using the `AcquisitionControlManager()` and also the "old" method using `acq = AcquisitionControl`. Also the artifact is present using an older commit like 984f4e3dfd4b890c910efb8a27734aeebf9720b6

The sequence was generated with these system specs:
```
system = Opts(
    rf_dead_time=20e-6,  
    rf_ringdown_time=30e-6,  
    grad_raster_time=1e-6,
    rf_raster_time=1e-6,
    block_duration_raster=1e-6,
    adc_raster_time=1e-6,
    max_grad=250e3 / GAMMA * 1e3,  # ~15 mT/m 
    grad_unit='mT/m', 
    max_slew=100, 
    slew_unit='T/m/s',
    B0=50e-3,
    gamma=GAMMA,
)
```

[tse_3d_image_artefact_74s_max_grad_250_no_elliptic.txt](https://github.com/user-attachments/files/21055980/tse_3d_image_artefact_74s_max_grad_250_no_elliptic.txt)

Here is a sample slice of the reconstruced image. The phantom is slightly visible in the background. The central, horizontal bright line stems from another issue.
[image]

The k-space looks heavily corrupted. 

[image]

Toreil, 2025-07-04:
Hi @h3lg3, could you post the k-space data? That might help me understand what's going on.

h3lg3, 2025-07-04:
Here the parameter for sequence creation:
```
fov = Dimensions(x=140e-3, y=140e-3, z=140e-3) 
enc = Dimensions(x=70, y=70, z=70) 
channels = Channels(ro='z', pe1='y', pe2='x') 
seq, header = tse_3d.constructor( 
    input_fov=fov, 
    input_enc=enc, 
    te=12e-3, 
    tr=500e-3, 
    etl=35, 
    n_dummies=0, 
    adc_bandwidth=10e3, 
    adc_oversampling=1, 
    g_ro_correction=90e-6, 
    rf_ex_duration=200e-6, 
    rf_ref_duration=200e-6, 
    rf_ex_phase=0, 
    rf_ref_phase=pi / 2, 
    trajectory=Trajectory.INOUT, 
    channels=channels, 
    system=system, 
) 
```

Toreil, 2025-07-04:
I will try these on the LUMC system, probably on Monday to see if I can reproduce the issue. Could you also try and run this on the RxData branch? The TSE example script should be able to run with the parameters you used here and reconstruct the data. I doubt it will change things since it’s mainly a data storage change but it’ll be an interesting tests since I agree with what you say about it being weird about which parameters work and which don’t.

Toreil, 2025-07-04:
I also had a weird combination of parameters which didn’t work, see issue #99

h3lg3, 2025-07-04:
I just tested the issue using the **TSE example sequence on main branch** with the same result.

[image]

```
params = {
    "echo_time": 20e-3,
    "repetition_time": 800e-3,
    "etl": 35,
    "trajectory": tse_3d.Trajectory.INOUT,
    "gradient_correction": 90e-6,
    "rf_duration": 200e-6,
    "fov": Dimensions(x=140e-3, y=140e-3, z=140e-3),
    "channel_ro": "z",
    "channel_pe1": "y",
    "channel_pe2": "x",
    "ro_bandwidth": 10e3,
    "n_enc": Dimensions(x=70, y=70, z=70),
}
```
```
    # time delay at the end of RF event, SETS RF DELAY!
    rf_dead_time=20e-6,
    grad_raster_time=1e-6,
    rf_raster_time=1e-6,
    block_duration_raster=1e-6,
    adc_raster_time=1e-6,
    rf_ringdown_time=0,
    max_slew=50,
    slew_unit="T/m/s",
    B0=50e-3,
    gamma=42.576e6
```

schote, 2025-07-15:
> Besides the bad image quality we see the central line artifact previously observed, but we also see a shift of the image, which might be due to a phase offset. The image shift is not present using the `acq control manager branch` but the `main branch`

Could you provide the commit IDs you're referring to? You can use `git rev-parse <branch-name>` to retrieve them. 
This will help identify which commit caused the issue.

schote, 2025-07-15:
> Hi! Since a few weeks I observe an image artifact for certain sequence parameters for the 3D TSE sequence. Mainly a large matrix size (70x70x70: ADCxPE1xPE2) combined with a low readout bandwidth (10 kHz). Weirdly the artifact disappears if I increase the bandwidth to 20 kHz (everything else unchanged). It also disappears if I change the matrix size to 70x70x10 with BW = 10 kHz (less PE2 steps). Therefore I dont think its a sequence issue. I dont think its an eddy current issue either because it is also present with with longer TE. Also present with BW 12 kHz. Not present with BW 15 kHz or 20 kHz. Not present with 30x30x30 matrix size. No artifact with 10x70x70, BW 10 kHz. The artifact is present for 10x70x70 and BW = 10e3/7. The artifact is present for 140x70x70 and BW = 20. In order to rule out a local hardware issues, I would be happy if someone could test the sequence on a different system. I attached the sequence as a .txt file. I tested the sequence on the current main branch [c17fd28](https://github.com/schote/nexus-console/commit/c17fd2865e9bd918ac85b78b66e71eb3f08d70dc) using the `AcquisitionControlManager()` and also the "old" method using `acq = AcquisitionControl`. Also the artifact is present using an older commit like [984f4e3](https://github.com/schote/nexus-console/commit/984f4e3dfd4b890c910efb8a27734aeebf9720b6)

Regarding the acquisition process `AcquisitionControlManager()` and `acq = AcquisitionControl()` are identical. The only difference is, that the managed version uses a separate process for the acquisition. Commit 984f4e3dfd4b890c910efb8a27734aeebf9720b6 already includes the removal of the phase reference signal. I could imagine that it is related to this, so it would be good to test commit 7311a0930e961601292bbd20e1c1348dc721d74a.

h3lg3, 2025-07-31:
@schote you proposed to test the artifact with commit https://github.com/schote/nexus-console/commit/7311a0930e961601292bbd20e1c1348dc721d74a
Unfortunately the artifact did not disappear with this commit. Maybe @Toreil could test the same sequence as posted above on his system to rule out any hardware issues on our side.

Toreil, 2025-07-31:
If you're on the scanner now could you test it with the RxCard_Update branch? There were some issues with properly grabbing the receive data from the cards. For some of these errors (such as those addressed in the RxCard_Update it's important to know a few of the parameters defined in your device_config file too (spcm_freq and number of enabled channels in that case), could you share you config file too? or the meta data stored in the data folder of the acquisition (if you're saving the data).

h3lg3, 2025-07-31:
I am not at the scanner currently, I can test it next week. Here is the meta-data from yesterday and the config file. 

[meta.json](https://github.com/user-attachments/files/21529733/meta.json)

[device_config.txt](https://github.com/user-attachments/files/21529762/device_config.txt)

h3lg3, 2025-08-06:
Hi @Toreil @schote ! The images look good using the RxCard_Update branch!! Amazing! This has been bugging me for a while, thanks for the fix. 

RxCard_Update branch [3b946f6](https://github.com/schote/nexus-console/commit/3b946f61809833da80b27a7405734747587a29ec)
[image]

Current main branch [bd4029c](https://github.com/schote/nexus-console/commit/bd4029c0b829747872fcfb538200760931afe80a)
[image]

[image]

Toreil, 2025-08-06:
Wow good to hear. I'm honestly surprised it makes such a difference but maybe the problem was more serious than I thought.

h3lg3, 2025-08-26:
This issues is solved with the new changes from PR #115 to main.

closed 2025-08-26 by h3lg3